projectweekend/Pi-GPIO-Server #1: Renaming GPIO pins in web UI

coopak, 2015-02-17:
After adding pins to the config file, they appear with a descriptive title of "{{ p.num }}: {{ p.mode }}" (e.g., 17: LOW). Is there any way to assign custom titles for each panel heading?

Pi-GPIO-Server/pi_gpio/static/partials/pins.html:
        <div class="panel-heading">
            <h3 class="panel-title">{{ p.num }}: {{ p.mode }}</h3>
        </div>

projectweekend, 2015-02-18:
Thanks for opening the issue. There is currently not a way to do this, but I have a couple of ideas for changes. One challenge is that there is no persistence layer for Pi-GPIO-Server, and that was intentional. I didn't want it to require setting up a database, or something like that, if someone wanted to use it. With that said, when editing something from the UI, there is no place to store these edits so that they survive restarts of the server application. 

What if I extended the configuration file format to allow you to give a pin a custom `name` property. You set the name you want in the config file when you initialize the pins. If a `name` is present, that's what will display on the page, else the current behavior would be default. Would this help solve your problem?

coopak, 2015-02-18:
That would be amazing. I don't think it adds unnecessary complexity to the installation of Pi-GPIO-Server but makes implementation and daily use much more user friendly.

projectweekend, 2015-02-21:
Hey! I just pushed up some changes to allow the optional `name` prop in the config file. README was updated with documentation. Check it out and let me know if there are any issues. Thanks!

coopak, 2015-02-23:
Looks perfect! I'll see how it goes updating this week...

coopak, 2015-03-08:
Works like a charm:
<a href="http://www.youtube.com/watch?feature=player_embedded&v=3EQfkEUouAg" target="_blank"><img src="http://img.youtube.com/vi/3EQfkEUouAg/0.jpg" 
alt="IMAGE ALT TEXT HERE" width="240" height="180" border="10" /></a>

closed 2015-03-08 by coopak

projectweekend, 2015-03-08:
Rock on! So glad this worked out for you.