projectweekend/Pi-GPIO-Server #2: cant run on RPi2

rodcar125, 2015-06-17:
Hi!

After install i just cant run it on my RPi2. When I start the program with python from the command line y get the following...

pi@S3RPi ~/Pi-GPIO-Server $ python pi_gpio_server.py 
Traceback (most recent call last):
  File "pi_gpio_server.py", line 1, in <module>
    from pi_gpio import app, socketio
  File "/home/pi/Pi-GPIO-Server/pi_gpio/__init__.py", line 16, in <module>
    import urls
  File "/home/pi/Pi-GPIO-Server/pi_gpio/urls.py", line 4, in <module>
    from handlers import PinList, PinDetail
  File "/home/pi/Pi-GPIO-Server/pi_gpio/handlers.py", line 3, in <module>
    from config.pins import PinHttpManager
  File "/home/pi/Pi-GPIO-Server/pi_gpio/config/pins.py", line 2, in <module>
    from .gpio import BaseGPIO
  File "/home/pi/Pi-GPIO-Server/pi_gpio/config/gpio.py", line 1, in <module>
    import RPi.GPIO as GPIO
RuntimeError: This module can only be run on a Raspberry Pi!

Is this normal due to lack of support for RpPi2 or did I messed up something im not aware of?

Thanks and great job!
Greetings,
Rodrigo C

projectweekend, 2015-06-17:
Thanks for the feedback! I only recently got a Pi2 so I haven't been able to test yet. That said, I dod a little research on your error and the problem is in the RPi.GPIO library. That library needs an upgrade on the Pi2 before it can work. Check this out article: [https://blog.adafruit.com/2015/02/05/how-to-fix-error-loading-rpi-gpio-python-library-on-your-brand-new-raspberry-pi-2/](https://blog.adafruit.com/2015/02/05/how-to-fix-error-loading-rpi-gpio-python-library-on-your-brand-new-raspberry-pi-2/). I will put a fix in the install script in next couple days so that this library gets upgraded as mentioned in this article. In the meantime, feel free to update it manually.

rodcar125, 2015-06-17:
Thanks for the response!
I just updated the GPI.GPIO library per the instructions on the link you provided and im happy to report that its working. thanks a lot!

Rod

projectweekend, 2015-06-19:
Awesome. Glad it worked out. I updated the `requirements.txt` file to get the latest version of `RPi.GPIO` to support the Raspberry Pi 2 going forward. Thanks again for pointing this out.

closed 2015-06-19 by projectweekend