projectweekend/Pi-GPIO-Server #4: Can't connet socket client to server

gamugamu, 2016-02-17:
Hi, 
This is a really nice API,

From your demo example, I can't connect to the server and listen to the event. Is there something special to do on the client side ? (the configuration web page work well at http://localhost:5000 on my web browser) 

But when I try to test a connection, I can't reach the socket. 

Here is your example:
```
var socket = io.connect( 'http://your_raspberry_pi.local' );
ç
// listen and receive data in the callback
socket.on( 'pin:list', function ( data ) {
  // do something with data
  console.log( data );
} );

// emit 'pin:list' to trigger a response with data
socket.emit( 'pin:list' );
```

Here is mine:

```
var io  require = require('socket.io-client');
var socket = io.connect('http://localhost:5000');

socket.on('connect'), function(data){
    console.log("connected");
});

socket.on('connect_error', function(error)){
  console.log("error : " + error);
 // trigger Error: xhr poll error
}
```

But the socket always trigger "Error: xhr poll error", because it can access at http://localhost:5000.
I have tried with http://raspberrypi.local:5000/ , http://raspberrypi.local, static private ip, nothing works.  tested on local computer, those url work on web browser, but on my client js code, I can never reach it.

On your example, you hint the code with 'http://your_raspberry_pi.local', but I clearly can't figure out how to reach the server socket.
 Thanks you for our help in advance.

projectweekend, 2016-02-17:
Thanks for opening the issue. I'm a bit slammed with my day job this week, but I'll get to this as soon as I can.

projectweekend, 2016-02-18:
One quick question, does the built-in demo page I included in the app work if you access it from another computer on the same network as your Raspberry Pi?

gamugamu, 2016-02-18:
Thank you for the quick answer.
Yes, your demo work nice remotely: I can access to the gpio functionality remotely. Tested on another computer and phone device from a wlan, and it worked like a charm. This is just the socket.io from a js  client code where I can't connect to the localhost at port 5000.

gamugamu, 2016-02-18:
Ok I found out, after debugging from your code I was getting a runtime error: ''You need to use a gevent-socketio server".

It appear to be an issue from flask-socketio:
https://github.com/miguelgrinberg/Flask-SocketIO/issues/39

So changed my version of socketio client to 0.9.16 (was 1.. something) and this did the trick.
`npm install socketio@0.9.16`

Now I can connect to the socket server.

closed 2016-10-05 by projectweekend

nabinkhadka, 2017-05-24:
I used [this](https://cdnjs.cloudflare.com/ajax/libs/socket.io/0.9.16/socket.io.min.js) and there is no problem