projectweekend/Pi-GPIO-Server #7: Mount failed for selinuxfs after running ./install.sh

RadeNarancic, 2016-10-03:
Hi,

I am getting the following error on boot after running the install.sh:

`Mount failed for selinuxfs on /sys/fs/selinux: No such file or directory`

I have tried this on Raspbian and Raspbian Lite, got the same error on both. Any advice?

projectweekend, 2016-10-05:
Thanks for opening an issue. What device are you using? I will do a little troubleshooting this weekend.

RadeNarancic, 2016-10-05:
Hi,

I am using Raspberry Pi 3 model B and the newest version of Raspbian.

brybad, 2016-10-19:
I am also having the same issue. It happens when I am using VNC to connect remotely with GPIO enabled and I use the shutdown now -r command. When using the GUI to shutdown, nothing happens when you click on shutdown or reboot. It has happened twice now and I will have to rebuild the C++ builds again...

I will try and go around the GPIO portion but will lose touchscreen functionality.

brybad, 2016-10-19:
Update to my previous post. I left GPIO disabled and still have the same issue. VNC was the only enabled add-on.

mdvarga, 2016-11-04:
Hi,

I was wondering if there have been any updates or investigations on this. I recently tried installing this as per the readme and I too encountered the following on reboot:

`Mount failed for selinuxfs on /sys/fs/selinux: No such file or directory`

I'm running Raspberry Pi 3 Model B with a completely fresh install of Raspbian Jessie Lite.

projectweekend, 2016-11-04:
Sorry guys, I haven't had time to investigate this yet. I wrote this long before the RPi3 was around. It may be something with the Upstart install? If anyone wants to try changing the script to remove Upstart and use Systemd instead that would be cool. I've been slammed at work and haven't had a chance to tinker with this.

brybad, 2016-11-04:
It does have to do with Upstart and the fact that Systemd is now included with the Raspian image. For some reason Raspian still installs with Init active and functional with Systemd not active but in the system. Once something is installed using Upstart, Systemd is then activated in Raspian for some reason and with no scripts present, it won't start up.

I haven't had a chance to post this information to Upstart yet as I haven't figured out if it is an Upstart issue or Raspian issue. In theory the Wheezy image from before March of this year (2016) should work without issues as it did not include Systemd.

brybad, 2016-11-04:
It may work to uninstall Systemd prior to installing Upstart. Otherwise [https://wiki.debian.org/Upstart](url) may help with the GRUB modification.

mdvarga, 2016-11-04:
I made some modifications locally and got everything to work. It is indeed upstart that is causing the failure. I created a gpio-server.service file for systemd to replace upstart:

```
[Unit]
Description=Pi GPIO Server
After=multi-user.target

[Service]
Type=idle
WorkingDirectory=/home/pi/Pi-GPIO-Server
ExecStart=/usr/bin/python /home/pi/Pi-GPIO-Server/pi_gpio_server.py

[Install]
WantedBy=multi-user.target
```

And I modified the install_system.py script to copy this new service configuration to:

`/lib/systemd/system/gpio-server.service`

So, the modified install_system.py looks like:

# Copy Upstart script
#subprocess.call(["cp", "./install/gpio-server.conf", "/etc/init"])

# Copy systemd configuration
subprocess.call(["cp", "./install/gpio-server.service", "/lib/systemd/system/"])
subprocess.call(["chmod", "644", "/lib/systemd/system/gpio-server.service"])

After rebooting, everything seemed to work. I verified that the service was indeed running with:

`sudo systemctl status gpio-server.service`

projectweekend, 2016-11-04:
Sweet! I had a feeling that might be it. Thanks for testing that. Want to open a PR with your fixes?

mdvarga, 2016-11-04:
Sure! I'll open a PR after work.

daniel-sullivan, 2016-11-24:
Good bug! In case anyone else comes across this, until an official push / pull is made, mdvarga's instructions above work well *but* BEFORE rebooting, please run:

```
sudo apt-get install systemd-sysv
```

I was watching through the install expecting this, and it looks like the actual issue is in the install that at some point a requirement is made on upstart which changes the system init system back to that. The command above sets it back to systemd which is what Jessie is expecting

daniel-sullivan, 2016-11-24:
Update: I needed to install on another device and was able to capture where it does its evil...

```
...
Setting up python-chardet (2.3.0-1) ...
Setting up python-colorama (0.3.2-1) ...
Setting up python-distlib (0.1.9-1) ...
Setting up python-six (1.8.0-1) ...
Setting up python-html5lib (0.999-3) ...
Setting up python-openssl (0.13.1-2) ...
Setting up python-ndg-httpsclient (0.3.2-1) ...
Setting up python-urllib3 (1.9.1-3) ...
Setting up python-requests (2.4.3-6) ...
Setting up python-setuptools (5.5.1-1) ...
Setting up python-pip (1.5.6-5) ...
Setting up python-pyasn1 (0.1.7-1) ...
Setting up python-wheel (0.24.0-1) ...
Processing triggers for python-support (1.0.15) ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
avahi-daemon is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  libjson0
Suggested packages:
  graphviz upstart-monitor
The following packages will be REMOVED:
  systemd-sysv
The following NEW packages will be installed:
  libjson0 upstart
0 upgraded, 2 newly installed, 1 to remove and 0 not upgraded.
Need to get 515 kB of archives.
After this operation, 1,825 kB of additional disk space will be used.
Get:1 http://mirrordirector.raspbian.org/raspbian/ jessie/main libjson0 armhf 0.11-4 [1,110 B]
Get:2 http://mirrordirector.raspbian.org/raspbian/ jessie/main upstart armhf 1.11-5 [514 kB]
Fetched 515 kB in 1s (313 kB/s) 
Selecting previously unselected package libjson0:armhf.
(Reading database ... 44182 files and directories currently installed.)
Preparing to unpack .../libjson0_0.11-4_armhf.deb ...
Unpacking libjson0:armhf (0.11-4) ...
Setting up libjson0:armhf (0.11-4) ...
dpkg: systemd-sysv: dependency problems, but removing anyway as you requested:
 init depends on systemd-sysv | sysvinit-core | upstart; however:
  Package systemd-sysv is to be removed.
  Package sysvinit-core is not installed.
  Package upstart is not installed.
```

fabialvarado, 2016-11-28:
Hi,

Thanks for the job. I have the same problem at the moment, but not find the file install_system.py for edit.

where can locate the file install_system.py?

mdvarga, 2016-11-28:
@fabialvarado: Pi-GPIO-Server/install/install_system.py

fabialvarado, 2016-11-29:
thanxs :)

amdroid007, 2017-02-21:
How do I make sure the gpio-server service starts automatically on boot? Right now I can start the service manually by typing sudo service gpio-server start and it starts up and works fine, but I would like to get it to start automatically when i reboot. I have the gpio-servicer.service in /lib/systemd/system and it is chmod to 644

thanks!
J

skurimski, 2017-03-03:
Was the installer issue ever corrected?  I recently downloaded and attempted to use it although it appears the issue is still there?  I created/copied the files as mdvarga suggested although the service is failing to start with a status code of 1.  Any suggestions?

bill-orange, 2017-03-05:
Same here. Did a pull request ever go in?

fhadj, 2017-03-05:
hello 
after having installed upstart then a C++ agent then rebooting the Pi
My Pi is stuck with this error message 
Mount Failed for selinuxfs on /sys/fs/selinux: no such file or directory.
nothing is happening then 
how can I recover from that ?  I can t go to recovering mode even when pressing the shift button

projectweekend, 2017-03-05:
Hey! I haven't had time to maintain this lately. Thins have been really
busy at work. The problem here is the install and usage of upstart. On
newer Pis this should use systemd instead. If anyone wants to submit a PR
that swaps out upstart for systemd I'd be happy to give a quick review and
merge. I wrote this a long time ago. Thanks!

bill-orange, 2017-03-05:
Submitting the PR is beyond my skill level right now. I tested with Raspian 3.2.17 using the instructions in this post.  I was able to reboot without failure, however, the service appeared to run, but I could not connect.  I got the following errors during install:

Downloading pytz-2014.3.zip (482kB)
    100% |¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦| 491kB 299kB/s 
Collecting six==1.6.1 (from -r requirements.txt (line 10))
  Downloading six-1.6.1-py2.py3-none-any.whl
Collecting wsgiref==0.1.2 (from -r requirements.txt (line 11))
  Downloading wsgiref-0.1.2.zip
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-zjua_qhk/wsgiref/setup.py", line 5, in <module>
        import ez_setup
      File "/tmp/pip-build-zjua_qhk/wsgiref/ez_setup/__init__.py", line 170
        print "Setuptools version",version,"or greater has been installed."
                                 ^
    SyntaxError: Missing parentheses in call to 'print'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-zjua_qhk/wsgiref/

fhadj, 2017-03-05:
in my side ...I have to use with systemd the 2 following scripts originally wrote to work with upstart 

script 1 
=====================================================
    sudo sh -c 'cat <<EOT > /etc/init/mtc_agent.conf
    description "MTConnect Agent"

    start on (net-device-up
              and local-filesystems
          and runlevel [2345])

    stop on runlevel[016]

    exec /usr/local/bin/agent run /etc/mtconnect/agent/agent.cfg
    respawn
    respawn limit 10 600
    EOT'

and this one 

script 2
=====================================================
    sudo sh -c 'cat <<EOT > /etc/init/mtc_adapter.conf
    description "MTConnect Adapter Simulator"

    start on (net-device-up
              and local-filesystems
          and runlevel [2345])

    stop on runlevel[016]

    exec ruby /etc/mtconnect/adapter/run_scenario.rb -l /etc/mtconnect/adapter/VMC-3Axis-Log.txt
    respawn
    respawn limit 10 600
    EOT'

any ideas ?

Thanks by advance

skurimski, 2017-03-06:
I will send a few dollars for coffee/soda money if anyone is willing to fix this.

brybad, 2017-03-06:
I haven't tried to uninstall systemd prior to rebooting. Unfortunately the MTConnect org has been extremely silent and no responses to their charter members. I may be able to try in a few days to compile MTConnect adapter and uninstall systemd.

fhadj, 2017-03-06:
FYI
I have configured my previous upstart scripts  like this 

pi@raspberrypi:/lib/systemd/system $ more mtc_adapter.service
[Unit]
Description= MTConnect Adapter Simulator
After=multi-user.target

[Service]
Type=idle
WorkingDirectory=/etc/mtconnect/adapter
ExecStart=/usr/bin/ruby /etc/mtconnect/adapter/run_scenario.rb -l /etc/mtconnect/adapter/VMC-3Axis-Log.txt
Restart=on-failure     
RestartSec = 60

[Install]
WantedBy=multi-user.target

 
pi@raspberrypi:/lib/systemd/system $ more mtc_adapter.service
[Unit]
Description= MTConnect Adapter Simulator
After=multi-user.target

[Service]
Type=idle
WorkingDirectory=/etc/mtconnect/adapter
ExecStart=/usr/bin/ruby /etc/mtconnect/adapter/run_scenario.rb -l /etc/mtconnect/adapter/VMC-3Axis-Log.txt
Restart=on-failure     
RestartSec = 60

[Install]
WantedBy=multi-user.target

pi@raspberrypi:/lib/systemd/system $ more mtc_agent.service
[Unit]
Description= MTConnect Agent
After=multi-user.target

[Service]
Type=idle
WorkingDirectory=/usr/local/bin
ExecStart=/usr/local/bin/agent run /etc/mtconnect/agent/agent.cfg
Restart=on-failure
RestartSec = 60

[Install]
WantedBy=multi-user.target

and now it works perfectly well !!! I can even see traffic between my agent and my MTconnect adapter !!

pi@raspberrypi:/lib/systemd/system $ systemctl status | grep mt
           │ ├─mtc_adapter.service
           │ │ └─1402 /usr/bin/ruby /etc/mtconnect/adapter/run_scenario.rb -l /etc/mtconnect/adapter/VMC-3Axis-Log.txt
           │ ├─mtc_agent.service
           │ │ └─1390 /usr/local/bin/agent run /etc/mtconnect/agent/agent.cfg
               │ ├─970 /usr/lib/gvfs/gvfs-mtp-volume-monitor
                 └─1474 grep --color=auto mt

Thanks mdvarga  !!!

brybad, 2017-03-07:
That is excellent news! Maybe I can get back to my project again in a couple of weeks.

sukima, 2017-03-08:
So the user story is this:

* Given a user who want a simple API to POST GPIO instructions to
* When the user installs this program via the `install.sh`
* And reboots
* Then the system is bricked with a missing selinuxfs error

This is a pretty serious bug for it to be addressed so casually. Curious what is so important about Upstart that this small python app must have it? Why is it required during install. I realize that for most of us this is par for the course but I can not imagine my 14 year old son trying this program only to be confronted with a destroyed system and without extensive Debian knowledge be able to recover and instead need to re-flash the entire OS.

The Upstart vs. SystemD vs. init debate is simply bike-shedding at the expense of a deplorable user experience. An app like this should offer out of the box support for the RasberryPi's default startup script with the alternatives being **optional** not required and most certainly not breaking the very fundamentals of the system to boot! No web server I know does that.

projectweekend, 2017-03-08:
I appreciate the candid feedback @sukima. I made this long time ago, for myself, and did not end up using it on Raspberry Pi beyond B+. This may have even been before jessie release too. I used Upstart because it's what I liked for managing long running processes at the time. 

No one is required to use the install script. It's just something that worked for me back when I made it. The Python web app is Flask and can be launched in any way you choose after cloning the code. The issue is solely in the install script. If it's that much of an issue I will simply remove that script and documentation pointing to it. I no longer have the time to maintain this by myself. That's the benefit of Github. Others can fork it, improve it, and submit to merge it back upstream. 

Finally, before anyone gets too upset, the included [license](https://github.com/projectweekend/Pi-GPIO-Server/blob/master/LICENSE) clearly states: "THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND".

sukima, 2017-03-08:
Apologies for the tone. I hooked up a PR to address that. Thank you for making it. I'm now using is successfully.

PR #8 (2017-03-09): Add systemd script to install
This also disables the upstart install that is breaking many RaspberryPis.

Closes #7
closed 2017-03-09 by projectweekend